Doodle3D/WiFi-Box #3: Test sending big files (>4MB) to Ultimaker

peteruithoven, 2016-04-01:
1. Update the box to the latest beta version (also `Remove sketches and settings`). 
1. Let the Doodle3D send more gcode; In Chrome or Firefox, open the `Web console`. Paste the following and press enter.  
`MAX_POINTS_TO_PRINT = Printer.MAX_GCODE_SIZE = 999999999999999999999;`  
Keep this Web console open. 
1. SSH into the WiFi-Box. 
1. Execute the following command `while true; do echo "$(cat /proc/meminfo | grep MemFree | sed -E 's/^[^0-9]+//') $(date '+%H:%M:%S')"; sleep 10; done`. (This will keep checking the available RAM). Keep this terminal window open while printing. You might want to increase the buffer of your terminal so it can "store" more results. 
1. Make a big drawing (lot's of text, extruded high)
1. Open in seperate browser tabs the following api endpoints (urls): 
  - `[WiFi-Box ip]/d3dapi/info/status`
  - `[WiFi-Box ip]/d3dapi/printer/progress`
1. In the doodle3d client press the Print button. 
1. Check the api endpoints during the print by refreshing the browser tabs. During the printing there should be `buffer_size` and `max_buffer_size` values. The `buffer_size` should increate, but stay under `max_buffer`. 
1. When the buffer is full (when the `buffer_size` is gets very close to `max_buffer_size`), in the browser tab with the Doodle3D client, in the Web console, there should appear a message containing `buffer_full`. When the buffer is full, messages containing `Printer:waitForBufferSpace` should appear. This should appear for about 30 seconds (because it keeps checking the buffer). Then when the buffer is 75% it should continue sending print parts. 
1. Make sure the terminal keeps recording the free memory. We're hoping it doesn't drop below 2000 kB. 
1. After te print copy the results to a file and sort this, include the file.

olijf, 2016-04-01:
Larger doodles makes the page crash in Google Chrome. 
[image]

If I try to create a doodle with total_lines > max_buffer_size the page crashes everytime.

olijf, 2016-04-01:
I tested a very large print on the Ultimaker 2, but stopped it halfway because it was to big. The free memory dropped 2 times below the magic barrier. Included are the logs.

[ultimaker print 1.zip](https://github.com/Doodle3D/WiFi-Box/files/199647/ultimaker.print.1.zip)

Also after stopping the print navigating away from the page will still result in this message being shown:
[image]

peteruithoven, 2016-04-01:
On the Chrome crash. This didn't happen the second attempt? It crashed at what moment? While drawing, starting sending, during sending etc?

olijf, 2016-04-01:
It crashed after i pressed print. Before that drawing was very sluggish. Tested on both my laptop (8gb ram) and the shuttle (4gb)
As it is fairly reproducible ill try to test it in other browsers too

peteruithoven, 2016-04-01:
Ah, and as long as the number of lines didn't go above 3145728 (the max buffer (in kb)) it cashes (when pressing print)?

olijf, 2016-04-01:
It crashed at around 1200000 lines so if 1 line of code is around 40 bytes (characters) it would be at 48000000 bytes or ~45MB

olijf, 2016-04-01:
2nd print on the ultimaker was succesfull it dropped to 1940 somewhere at the start. Dont know why. 
[ultimaker print 2.zip](https://github.com/Doodle3D/WiFi-Box/files/199848/ultimaker.print.2.zip)

woutgg, 2016-04-01:
What should be taken into account while looking at free memory is that log rotation may kick in with 2 minute intervals. This leads to a brief time of less free memory (and increased CPU usage) due to copying and compressing the log file.

olijf, 2016-04-04:
## Print 3 from tablet (0.10.10-c)
During my print from my Nexus 9 the ultimaker started making strange noises so @mesut-pehlivan powered it off. I hope this is a printer issue. Can we check this in the logs?

[ultimaker tablet print.zip](https://github.com/Doodle3D/WiFi-Box/files/202542/ultimaker.tablet.print.zip)

peteruithoven, 2016-04-04:
@olijf Could we try this once more, since the first print was manually interrupted?

olijf, 2016-04-05:
## Print 4 from tablet (0.10.10-c)
Again same effect. 

[ultimaker tablet print 2.zip](https://github.com/Doodle3D/WiFi-Box/files/204234/ultimaker.tablet.print.2.zip)

olijf, 2016-04-05:
It looks like the wifibox disconnects while the tablet is sending gcode en then the wifibox reconnects but the sending of gcode is never resumes, so the wifibox eventually runs out of gcode to print en then just stops.

peteruithoven, 2016-04-05:
@olijf is right: https://github.com/Doodle3D/doodle3d-client/issues/304

Let's do a extra test from a computer with a WiFi-Box connected through ethernet cable, to see if there are other issues.

olijf, 2016-04-05:
## Print 5 from PC (0.10.10-c)
Printer ran out of gcode to print. Does not look like it is the same issue. 

```
Message:set:  Sending doodle to printer: 12% notice false true
printer:sendPrintPart response:  Object {data: Object, msg: "could not add gcode (buffer_full)", status: "fail"}data: Objectmsg: "could not add gcode (buffer_full)"status: "fail"__proto__: Object
71```
[ultimaker print 5.zip](https://github.com/Doodle3D/WiFi-Box/files/204384/ultimaker.print.5.zip)

olijf, 2016-04-05:
## Print 6 from PC using ethernet (0.10.10-c)
Print failed. hanged at 71% I do not understand why as the ultimaker did have enough buffer to continue. This might be a different issue?
[ultimaker print 6.zip](https://github.com/Doodle3D/WiFi-Box/files/204716/ultimaker.print.6.zip)

peteruithoven, 2016-04-20:
We've released a new beta version: 0.10.10-d. ([Release notes](http://doodle3d.com/releasenotes/))
Let's try these tests again. Maybe twice from a desktop and twice from a tablet. I think we noticed that especially on a tablet the wifi connection was problematic.

olijf, 2016-04-20:
## Test from tablet using 0.10.10-d
I tested the new fix to see if it helps... unfortunately the print stopped halfway but according to @peteruithoven this is not something related to https://github.com/Doodle3D/doodle3d-client/issues/304
it seems to be a buffer issue. 
I cleared my cache and restarted the wifibox and repeated the test but it would not help.
[0.10.10-d fix.zip](https://github.com/Doodle3D/WiFi-Box/files/228205/0.10.10-d.fix.zip)

Also after restarting the printer because it hanged I experienced a lot of ```[ 1514.270000] usb 1-1: device descriptor read/64, error -145
[ 1538.760000] usb 1-1: new full-speed USB device number 14 using ehci-platform
``` errors. seems something is wrong while communicating with the ultimaker. Also print3d does not get started. This seems to be something different... I'll look into it tomorrow.

olijf, 2016-04-21:
## Test from tablet 0.10.10-d
Tested using my tablet.. unsuccessful. 
After recieving an ``Printer:sendPrintPart: failed (AJAX status: 'timeout') AJAX exception (if any): timeout `` It did resume sending the data to the wifibox but recieved a `` Printer:sendPrintPart: unexpected failure response for API endpoint printer/print (seq_num_mismatch)`` error.
After this it never resumed buffering. Printer stopped printing when the buffer was depleted.
[tablet print 2 0.10.10-d.zip](https://github.com/Doodle3D/WiFi-Box/files/229714/tablet.print.2.0.10.10-d.zip)

woutgg, 2016-04-21:
That is strange as the lines printing that error have been modified to include the specific sequence numbers ([here](https://github.com/Doodle3D/doodle3d-client/blob/develop/js/Printer.js#L248-L249)). After checking the js in the image, it seems it  still contains old javascript.

peteruithoven, 2016-04-22:
For some reason 0.10.10-d indeed contained old javascript code. We've released 0.10.10-e and this does contain the latest javascript code. Let's try the tests from https://github.com/Doodle3D/WiFi-Box/issues/3#issuecomment-212392384 again. 
Let's also download the logs after an issue occurs and then reboot the whole WiFi-Box. This way the logs should only contain the issue at hand.

olijf, 2016-04-22:
## Test from tablet 0.10.10-e
Tested using my tablet.. unsuccessful. 
It hanged while waiting for the buffer to drop below 75% which somehow never happened after a certain time.
[0.10.10-e.zip](https://github.com/Doodle3D/WiFi-Box/files/231417/0.10.10-e.zip)

olijf, 2016-04-22:
## Test from tablet 0.10.10-e
Again same buffer issue: hangs while waiting for the buffer to drop below 75%
[0.10.10-e test 2.zip](https://github.com/Doodle3D/WiFi-Box/files/231439/0.10.10-e.zip)

olijf, 2016-04-22:
## Test from tablet 0.10.10-e 3
Again same buffer issue.
[0.10.10-e print 3.zip](https://github.com/Doodle3D/WiFi-Box/files/231496/0.10.10-e.print.3.zip)

olijf, 2016-04-22:
## test from pc 0.10.10-e 1
Test successful 
[0.10.10-e test from pc.zip](https://github.com/Doodle3D/WiFi-Box/files/231604/0.10.10-e.test.from.pc.zip)

olijf, 2016-04-22:
## Test from pc 0.10.10-e 2
Test succesful
[0.10.10-e print 2.zip](https://github.com/Doodle3D/WiFi-Box/files/231605/0.10.10-e.print.2.zip)

olijf, 2016-04-22:
I looked into why the ultimaker usb did strange things (read error -145 means timeout btw) and according to  openwrt this is a known issue with certain usb devices when they draw too much power from the port on atherors hardware. see: https://dev.openwrt.org/ticket/16467

olijf, 2016-04-25:
## Test from galaxy tab tablet 0.10.10-e 1 
This time I used another tablet to see if the buffer issues would arise. but this test was successful. I am unsure that this has anything to do with the tablet at hand. I did completely reset the wifibox and removed all settings just to have a fresh start.

olijf, 2016-04-25:
## Test from galaxy tab tablet 0.10.10-e 2
I did another test, this time I didn't enable the heated bed. Shape did not stick very well so I stopped the print.
After I pressed print again the wifibox hung in the buffering state much like printing from my nexus 5.
Also print3d did stop running.
[0.10.10-e print 2 galaxy tab.zip](https://github.com/Doodle3D/WiFi-Box/files/234267/0.10.10-e.print.2.galaxy.tab.zip)

peteruithoven, 2016-04-28:
@olijf Could you do 2 more of the same tests with your own tablet and adding a usb hub between the WiFi-Box and the 3D printer?

olijf, 2016-04-29:
Test from nexus 9 tablet 0.10.10-e
I tested using a small usb hub and verified that it worked correctly. 
Both prints where successful 
[nexus 9 0.10.10-e print.zip](https://github.com/Doodle3D/WiFi-Box/files/242755/nexus.9.0.10.10-e.print.zip)

[nexus 9 0.10.10-e test 2.zip](https://github.com/Doodle3D/WiFi-Box/files/242756/nexus.9.0.10.10-e.test.2.zip)

It might be possible that the usb ``-145`` issue is solved using a USB hub although it is not observed often.

olijf, 2016-04-29:
## Test from galaxy tab tablet 0.10.10-e
I did some more tests on the Galaxy tab. I forgot to upload the logs. If I remember correctly both were successful. Might be useful . 
[0.10.10-e print 3 galaxy tab.zip](https://github.com/Doodle3D/WiFi-Box/files/242761/0.10.10-e.print.3.galaxy.tab.zip)
[0.10.10-e print 4 galaxy tab.zip](https://github.com/Doodle3D/WiFi-Box/files/242762/0.10.10-e.print.4.galaxy.tab.zip)

peteruithoven, 2016-05-25:
We've released 0.10.10, so I'm closing this test.

closed 2016-05-25 by peteruithoven