Doodle3D/WiFi-Box #5: Test stopping big file (>4MB) to Wanhao

peteruithoven, 2016-04-01:
1. Update the box to the latest beta version (also `Remove sketches and settings`). 
1. Through the Doodle3D client settings window change the Log level to `Bulk`.
1. Let the Doodle3D send more gcode; In Chrome or Firefox, open the `Web console`. Paste the following and press enter.  
`MAX_POINTS_TO_PRINT = Printer.MAX_GCODE_SIZE = 999999999999999999999;`  
Keep this Web console open. 
1. Make a big drawing (lot's of text, extruded high)
1. Try the following 4 times: 
  1. In the doodle3d client press the Print button. 
  1. First two times: Press stop while still sending print.  
Last two times: press stop after finished sending. 

After te print (fails):
  - Create a comment in this issue
  - Include the firmware version. 
  - Copy the terminal contents to a file and sort this, include this file. 
  - Export the logfiles. You can attach them to a comment by first compressing them to a zip file. 
  - Copy the contents of the Web console and include this in a file. (Usually there is a right mouse button option to export to file).
  - When print failed, try to include the sketch (In the Settings window, press `Download SVG`). 

Let's try this 3 times. After a first print without issues we can skip steps 7, 9 and 10.

olijf, 2016-04-01:
[Wanhao stopping.zip](https://github.com/Doodle3D/WiFi-Box/files/199888/Wanhao.stopping.zip)

It seems the Wanhao never stops, even with smaller prints

peteruithoven, 2016-04-01:
(It does stop printing, but it doesn't seem to execute the end gcode)

olijf, 2016-04-01:
Yes, although according to the logs it looks like the ending gcode is actually sent to the printer!

woutgg, 2016-04-01:
Just for reference, there is an issue about this observance: https://github.com/Doodle3D/print3d/issues/40.
Also note that the currently used printer buffer space can be seen in the print3d logs.

woutgg, 2016-04-01:
I noticed that we now have logs of exactly which bytes are being written to the printer, so maybe with the knowledge of how much is data is left in the buffer we could figure out the actual bytes and run that through the s3g decompiler to find out where it gets stuck?

peteruithoven, 2016-04-04:
@olijf Could you try this once, with the stable firmware, with a regular sized sketch? I'd like to check / confirm if this changed with this release or this was a existing bug.

olijf, 2016-04-04:
## Print using non-beta 0.10.10 
I tested the Wanhao using the normal firmware just to test if this wasn't an issue. It seems the stop function works properly on this firmware. 

[wanhao stopping regular firmware.zip](https://github.com/Doodle3D/WiFi-Box/files/202581/wanhao.stopping.regular.firmware.zip)

peteruithoven, 2016-04-05:
@olaf you tested this with the beta a couple of times, but didn't include the log files?

olijf, 2016-04-05:
@peteruithoven If I am not mistaken these are all included in the zip file

peteruithoven, 2016-04-05:
@olijf maybe I'm not looking correctly but I'm only finding one `wifibox-log`s in the `Wanhao stopping.zip` file.

peteruithoven, 2016-04-18:
Also see: #40

peteruithoven, 2016-04-18:
@olijf There migth be a issue with the Makerbot end-gcode. Could you try commenting the following lines from the makerbot end gcode and trying to stop a print again? Please try commenting these one by one and attempting to stop the printer untill it works. If it starts stopping properly please try reenabling some lines again. Between each test reboot the WiFi-Box and 3D printer, this is to prevent code sticking in buffers. 
- `M72 P1  ( Play Ta-Da song )`
- `M137 (build end notification)`
- `M73 P100`
- `M73 P100 (end  build progress )`

You can comment by adding a `;` at the beginning of the line. 
Please try this with the latest beta release. (I'm not sure why this would only affect our beta.)

jeroensomers, 2016-04-19:
@peteruithoven
I commented the following two lines:

- `M73 P100`
- `M73 P100 (end  build progress )`

After this the print stops normally like it should.

closed 2016-05-02 by olijf