Doodle3D/WiFi-Box #10: Let's write documentation about the Doodle3D WiFi-Box in the Github wiki of this repository.

companje, 2016-05-21:
https://github.com/Doodle3D/WiFi-Box/wiki

companje, 2016-05-21:
This is a drawing explaining how the WiFi-Box connects drawing apps on different devices over WiFi to 3D-printers through USB.
[image]

companje, 2016-05-21:
This image explains shows how a drawing made in the Doodle3D app is send to the WiFi-Box which is connected through USB to a 3D-printer.

[image]

peteruithoven, 2016-05-21:
Most docs currently live here: 
http://www.doodle3d.com/help/wiki/firmware
http://www.doodle3d.com/help/wiki/serverside

This issue is about moving such docs to this repo's wiki?

companje, 2016-05-21:
Yes, eventually I would like to move all documentation from http://www.doodle3d.com/help/wiki to the wikis of the corresponding repository.