Doodle3D/WiFi-Box #11: Test status info on Ultimaker2's

peteruithoven, 2016-05-25:
With the following pull request Ultimaker2's (2go, extended etc, plus etc) should show status info 'Doodle3D heat up...', 'Doodle3D printing...' and 'Doodle3D done! on the display'. 
https://github.com/Doodle3D/doodle3d-firmware/pull/65
Let's test this with our Ultimaker 2 and Ultimaker 2 go. Try making two prints after eachother with at least one of those printers. Check that this doesn't break anything, like the regular menu's. 
Also check that this 'Doodle3D done!' shows up when stopping the print.

jeroensomers, 2016-06-14:
> With Ultimaker 2 try making two prints (without rebooting printer or WiFi-Box):

Done
> It shows the above mentioned statuses?

Yes on the UM2 it shows al three statuses.
> This doesn't break anything (like the regular menu's)?

No everything works fine
> With Ultimaker 2 go try making one print

Done
> It shows the above mentioned statuses?

No it shows "Printing with USB..."

> This doesn't break anything (like the regular menu's)?

No everything works fine

> With Ultimaker 2 go try making one print but stopping the print. Does it display 'Doodle3D done!'? 

No it doesn't say 'Doodle3D done!" it says "Printing with USB..."

peteruithoven, 2016-06-15:
With the beta version, in the settings window, could you try pressing the Restore settings to defaults button? This should retrieve the latest defaults and enable this feature. Could you then try the 2go test again? More info: https://github.com/Doodle3D/doodle3d-firmware/pull/65#issuecomment-226145239

jeroensomers, 2016-06-15:
YES,

When I pressed the "Restore settings to defaults" button, all settings where restored and the display says all three sentences (heating.. printing... done...)

The start and end gcode where not refreshed within the box, but now they are.

peteruithoven, 2016-06-15:
Awesome

closed 2016-06-15 by peteruithoven