Doodle3D/WiFi-Box #12: Test feature/button switch wificonfig

peteruithoven, 2016-05-25:
- [ ] wifi related config is correctly changed after pressing reset button? 
- [ ] Failsave still works? 
- [ ] Can we still call d3dapi from commandline?

peteruithoven, 2016-05-25:
> It becomed properly names access point?

yep

> Wifi related config is correctly changed after pressing reset button? 

`/etc/config/wireless` went from:
```
config wifi-device 'radio0'
	option type 'mac80211'
	option hwmode '11g'
	option path 'platform/ar933x_wmac'
	option htmode 'HT20'
	option disabled '0'

config wifi-iface
	option disabled '0'
	option network 'wlan'
	option encryption 'psk2'
	option device 'radio0'
	option mode 'sta'
	option key '...'
	option ssid 'Vechtclub XL F1.19'

config wifi-iface
	option network 'wlan'
	option encryption 'none'
	option device 'radio0'
	option mode 'ap'
	option ssid 'Doodle3D-4CC28C'
	option disabled '1'

config wifi-iface
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option ssid 'OpenWrt'
	option encryption 'none'
	option disabled '1'
```
To:
```
config wifi-device 'radio0'
	option type 'mac80211'
	option hwmode '11g'
	option path 'platform/ar933x_wmac'
	option htmode 'HT20'
	option disabled '0'

config wifi-iface
	option network 'wlan'
	option encryption 'none'
	option device 'radio0'
	option mode 'ap'
	option ssid 'Doodle3D-4CC28C'
	option disabled '0'
```
Which seems okay. 

> Failsave still works? 

Yes

> Can we still call d3dapi from commandline? 

Yes

closed 2016-05-25 by peteruithoven

olijf, 2016-05-25:
@peteruithoven maybe we also need to test the new API entry?

peteruithoven, 2016-05-25:
Good point

peteruithoven, 2016-05-25:
> After pressing button we can connect to it properly?

Yes

> After reboot we can connect to it properly?

Yes

> /d3dapi/network/reset also creates access point? 

Using the Chrome app Postman I tried this and it created the access point.

olijf, 2016-05-25:
ah ok, but nothing is displayed?

peteruithoven, 2016-05-25:
I didn't get a response, but then again I didn't test this using the ethernet cable, so that makes sense right?