Doodle3D/WiFi-Box #13: Test feature/always available

peteruithoven, 2016-05-25:
- [ ] Does it still connect to WiFi networks? 
- [ ] When the wrong password is given it becomes an access point? Does it have the proper name (e.g. Doodle3D-4CC28C)
  - [ ] Try this on desktop computer using connect.doodle3d.com. Does it give some useful feedback?
  - [ ] When you then give the right password it does connect properly? 
  - [ ] Try this on desktop computer using the settings popup. Does it give some useful feedback? 
  - [ ] When you then give the right password it does connect properly? 
  - [ ] Try this on a tablet / mobile phone. The tablet / phone will probably switch back to a previous wifi and not give feedback? 
  - [ ] When you connect to created access point and try this again with the right password it does connect properly?
- [ ] When the wrong ssid is given it becomes an acccess point? How: Create hotspot with mobile, select as target network, disable hotspot, attempt to connect to network. 
  - [ ] Try this on desktop computer using connect.doodle3d.com. Does it give some useful feedback?
  - [ ] When you then enable the hotspot again, can it connect properly? 
- [ ] When a wrong password is given and it became a access point and then the WiFi-Box is rebooted, does it become that access point again? 

In comment, mention specific WiFi-Box version. Tick / check all successfully completed test above. Quote all completed tests and share result behind each test. for example: 

> Does it still connect to WiFi networks? 

Yes, without any problems. 
> When the wrong password is given it becomes an access point? 
>   Try this on desktop computer using connect.doodle3d.com. Does it give some useful feedback?

It became an access point (Doodle3D-4CC28C) and it indicated: "Failed connecting to WiFi-network".

jeroensomers, 2016-05-30:
>Does it still connect to WiFi networks? 

Yes, without any problems

jeroensomers, 2016-05-30:
>When the wrong password is given it becomes an access point? Does it have the proper name (e.g. Doodle3D-4CC28C) 

It became an access point (Doodle3D-75BFD0) but I got no indication: "Failed connecting to WiFi-network".

jeroensomers, 2016-05-30:
> When you then give the right password it does connect properly?

Yes, without any problems.

jeroensomers, 2016-05-31:
> Try this on a tablet / mobile phone. The tablet / phone will probably switch back to a previous wifi and not give feedback? 

It didn't switch back when I gave the wrong password, the site gave me the following message: `Something went wrong, please try again`

jeroensomers, 2016-05-31:
>  When you connect to created access point and try this again with the right password it does connect properly?

Yes, without any problems

jeroensomers, 2016-06-07:
>When the wrong ssid is given it becomes an acccess point? How: Create hotspot with mobile, select as target network, disable hotspot, attempt to connect to network.

>Try this on desktop computer using connect.doodle3d.com. Does it give some useful feedback?

It works but it does not give any feedback.

>When you then enable the hotspot again, can it connect properly?

Yes, it connects properly

jeroensomers, 2016-06-07:
> When a wrong password is given and it became a access point and then the WiFi-Box is rebooted, does it become that access point again? 

Yes, it becomes a accesspoint again.

peteruithoven, 2016-06-15:
> > Try this on a tablet / mobile phone. The tablet / phone will probably switch back to a previous wifi and not give feedback? 

> It didn't switch back when I gave the wrong password, the site gave me the following message: `Something went wrong, please try again`

This is great, I expected the device to switch. We should try this a couple of times, just because it's good to know how it behaves. But for now it's tested and the changes seem release ready.

closed 2016-06-15 by peteruithoven