Doodle3D/WiFi-Box #14: Test improved uhttpd config

peteruithoven, 2016-05-25:
https://github.com/Doodle3D/doodle3d-firmware/pull/61 improved the uhttpd config to handle more requests. 

With the latest stable (0.10.10):
- Have the Doodle3D client open on multiple computers (preferably 5 or more) (use desktop, raspberry pi, tablets, phones etc).
- Also run the latest Cura (with the Doodle3D plugin enabled) on a computer. 
- send a large 3D print (something you can replicate)
- Keep an eye on: 
  - "Connection lost" warnings
  - Whether the print can be send successfully and how long this takes. 
- Update to the latest beta (0.10.11-...).  
- Try this again with a similar drawing. 

- [ ] Do we see this "Connection lost" message less? 
- [ ] Does sending the print roughly take an less or equal amount of time?
- [ ] Does the printer print normally? It doesn't for example stutter (halts printing for a short moment) sometimes?

jeroensomers, 2016-06-08:
> Do we see this "Connection lost" message less? 

The "Connection lost" message did not come. 

> Does sending the print roughly take an less or equal amount of time?

It didn't take any longer with the latest beta (0.10.11-a)

> Does the printer print normally? It doesn't for example stutter (halts printing for a short moment) sometimes?

Yes the printer printed normal there where no interruptions during the print process.

peteruithoven, 2016-06-08:
One remarkable note; we didn't experience "connection" issues (seeing "Connection lost") when using 5? computers with Doodle3D client + Cura with Doodle3D plugin with the latest stable, that didn't even have the improvements. Even though our WiFi situation isn't exactly stable.

jeroensomers, 2016-06-14:
I did a big test with multiple pc/laptops/tablets/phones/macs, I opened multple tabs so the Doodle3D drawing interface was opened more then 25 times (a average classroom number) 

First I did this test with the stable version (0.10.10)  these where my findings:

 - The printer was interrupted multiple times, this you can see back in the print.
 - the average response time was more then  400 milliseconds.

The second test I did withe the new beta release (0.10.11-a)

 - The printer was not interrupted.
 - The average response time was around 200/250 milliseconds.

In my opinion, the new beta release is more stable then the already 'stable' release.

peteruithoven, 2016-06-14:
Nice! 
In both versions the response time didn't raise to such a level that they triggered the "Connection lost" messages right?

jeroensomers, 2016-06-14:
Correct, in both versions I didn't get the message "Connection lost".

peteruithoven, 2016-06-15:
Thanks Jeroen, this seems tested enough.

closed 2016-06-15 by peteruithoven