Doodle3D/WiFi-Box #16: Disable pre-heating by default

peteruithoven, 2016-07-01:
Currently we pre-heat printers their nozzle and bed so that in a demo context users can print rapidly. But maybe this shouldn't be the default?

companje, 2017-06-20:
I suggest to remove pre-heating and cooling down to 'pre-heat temperature' from the default start and end-code. If pre-heating is still considered useful I think it should be a custom gcode call done by the Sketch app.

peteruithoven, 2017-06-21:
Why? 
That would also require a separate gcode command send after a print was completed? At which point there is a change the client isn't connected.

companje, 2017-06-21:
I mean for safety it would be better to cool down to 0 just like other slicers do.
If the app is still open it can re-preheat again if the user is still active.

peteruithoven, 2017-06-22:
I would suggest disabling the `Auto preheat` setting by default. This should result in the old sketch app not requesting a preheat. 
And since the end gcode doesn't listen to this setting I would also set the default preheat temperatures to 0. 

This means people can still enable preheat and we don't need to change logic, we only need to change the default settings.