Doodle3D/WiFi-Box #19: WiFi-Box unusable after update

peteruithoven, 2016-10-19:
We've had complaints recently that after trying to update the WiFi-Box is unusable. The first, third and last light do go on. It's possible to login with failsafe.

peteruithoven, 2016-10-19:
It's quite tricky to debug, since logread doesn't work. I can boot ubus (`ubusd`) and `logd` and then check logread, but then it doesn't contain anything yet.

companje, 2016-10-19:
[image]

[image]

[image]

[image]